bitpool/edge-bacnet #1: HOMEASSISTANT - NODERED

lucaskhourybr, 2023-06-30:
O nó Gateway não apresenta as Network Interface. 
Alguma solução ?

dblanchie, 2023-07-02:
> O nó Gateway não apresenta as Network Interface.
> Alguma solução ?

Please send through OS details, did you restart node red after deploying gateway node?

lucaskhourybr, 2023-07-02:
Estou rodando o node Red no sistema HAOS ( HomeAssistant ) . Já reinicei o NODERED e já reiniciei o servidor HAOS , porém sem sucesso.

lucaskhourybr, 2023-07-07:
Alguem tem alguma solução ja tentei diversas vezes porem sem sucesso

dblanchie, 2023-07-07:
> Alguem tem alguma solução ja tentei diversas vezes porem sem sucesso

I recommend running nodered on a seperate instance, we have tested on Ubuntu Server and Windows to date. You can grab the entire docker container from the link in the repo

lucaskhourybr, 2023-07-07:
Precisaria utilizar na plataforma no HOMEASSISTANT devido as integrações que estou criando .
Em instancia separada funciona mesmo conforme comentando .

bitpool-dev, 2023-07-18:
Hi @lucaskhourybr  - we currently do not support Home assistant os. Please use node-red directly on linux or windows, or docker for linux.

closed 2023-07-18 by bitpool-dev

gnlgrim, 2024-05-25:
For anyone interested. I got this working .
I am busy writing up a documentation for the whole process.
But in a quick way.    do everything as normal, but access node red through Home assistant IP and the port under the add-on configuration settings ( usually 1880 ) please remember to add a username and password.
You have to choose All interfaces under interface settings and set Broadcast Address to 255.255.255.255

Hope this helps. 
Will send everything over to devs when done so they can add it to documentation

lucaskhourybr, 2024-07-15:
Como que você realizou isso ainda estou com mesmo problema 2024 homeassistant/nodered não consigo visualizar as portas de rede.
[image]

gnlgrim, 2024-07-15:
you cannot access node red from inside home assistant.
for some odd reason if you do it this way home assistant blocks interface
You have to access it from node red external IP.
To check the port . go into home assistant 
settings
add ons
node red.
configuration
scroll to bottom and check port settings
""node red documentation can help with this ""
then type home assistant ip address followed by node red port
example:  192.168.1.113:1880
then you will be able to pick up Network interface

bitpool-dev, 2024-07-15:
Can you try with the bacnet server enabled? There is currently a bug with version 1.4.1 where the node stops / network interfaces disappear if the bacnet server is not enabled ? This will be resolved with version 1.4.2

lucaskhourybr, 2024-07-19:
Boa noite.

Estou utilizando o IP e a porta 1880 , porém mesmo assim ainda nãe me apresenta as placas de rede, conforme imagem abaixo

[image]

Além disso ja habilitei o servidor bacnet conforme imagem abaixo e reinciei ele porém sem sucesso ainda.

[image]

bitpool-dev, 2024-07-23:
Are you able to try again with the version 1.4.2 ?

lucaskhourybr, 2024-08-03:
@bitpool-dev mesmo na versão 1.4.3 atual não funciona infelizmente, não aparece a porta de rede.