bitpool/edge-bacnet #2: missing edge-bacnet-datastore.cfg

ravz, 2023-07-02:
I'm seeing this error on first install of the node

```
Read_Config_Sync error: Error: ENOENT: no such file or directory, open 'edge-bacnet-datastore.cfg'
    at Object.openSync (node:fs:601:3)
    at Object.readFileSync (node:fs:469:35)
    at Read_Config_Sync (/home/rav/.node-red/node_modules/@bitpoolos/edge-bacnet/common.js:232:15)
    at new BacnetClient (/home/rav/.node-red/node_modules/@bitpoolos/edge-bacnet/bacnet_client.js:33:41)
    at new BitpoolBacnetGatewayDevice (/home/rav/.node-red/node_modules/@bitpoolos/edge-bacnet/bacnet_gateway.js:78:27)
    at Object.createNode (/home/rav/.nvm/versions/node/v18.16.1/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/flows/util.js:90:27)
    at Flow.start (/home/rav/.nvm/versions/node/v18.16.1/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:206:48)
    at start (/home/rav/.nvm/versions/node/v18.16.1/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/flows/index.js:394:33) {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: 'edge-bacnet-datastore.cfg'
}
```

dblanchie, 2023-07-02:
Make sure if you where running an existing version to completely delete all existing nodes. We recently implemented import export function for storing bacnet database

bitpool-dev, 2023-07-17:
Hi @ravz  - this error indicates no config file was found. It creates a new file if this error is caught. Error msg will be removed on next node publish. You can ignore it.

closed 2023-07-17 by bitpool-dev