bitpool/edge-bacnet #3: "Read" no longer work after node-red restart

DSt00, 2023-07-03:
Hi,

While testing the flow following the youtube tutorial, the read function no longer works properly after restart of the node-red.

The gateway is finding my devices
[image]
but the read function is empty, even though it says 2 devices found
[image]

Any ideas on how to troubleshoot?
I'm currently testing it on Win10 virtual machine before implementing it on an edge device. After the issue occur I tried it on freshly installed node-red on my local machine, but there it cannot even find the devices.
[image]

dblanchie, 2023-07-10:
Conversation moved to discord for diagnosis will update in comments when resolved

bitpool-dev, 2023-07-18:
Issue surrounding hyper v installation. A solution was found via discord chat. Closing this case.

closed 2023-07-18 by bitpool-dev