bitpool/edge-bacnet #7: Write fails: pointsToWrite is empty

NGDWalker, 2023-08-11:
[image]

dblanchie, 2023-08-19:
interestingly we tested on BACnet IP Distech ECY VAV, we will test MSTP write functionality and look to resolve the issue, we had only tested on MSTP read to date. End of august we will be pushing more updates and functionality on this.

NGDWalker, 2023-08-22:
The error is in bacnet_write.html in addPointClicked @line 190. Seems like parentDevice is undefined:

TypeError: Cannot read properties of undefined (reading 'Heat_Valve_AO_7')
    at Proxy**.addPointClicked (<anonymous>:190:56)**
    at onClick (eval at iu (http://localhost:1880/resources/@bitpoolos/edge-bacnet/vue.global.prod.js:1:119793), <anonymous>:179:37)
    at Bt (:1:14252)
    at Lt (:1:14330)
    at HTMLButtonElement.n (http://localhost:1880/resources/@bitpoolos/edge-bacnet/vue.global.prod.js:1:63670) 
    {stack: 'TypeError: Cannot read properties of undefine…oolos/edge-bacnet/vue.global.prod.js:1:63670)', message: 'Cannot 
    read properties of undefined (reading 'Heat_Valve_AO_7')'}

NGDWalker, 2023-08-22:
I got further by changing : let key = `${device.address}-${device.deviceId}`;
to let key = `${device.address.address}-${device.deviceId}`;

I'll wait for your end of August push

bitpool-dev, 2023-08-23:
Hi @NGDWalker 
Thanks for your feedback, we are aware of this issue and will aim to fix it as part of our next node deployment. It looks like a UI bug for MSTP devices in the write node. 

Did the write operation work after you made the above code change ?

NGDWalker, 2023-08-23:
It does populate the "pointsToWrite", and does not throw. But did not actually write to the point.
Also, I reverted my edits and changed the test to an IP device with the same results (point not written).

One possible issue, parsing the payload always as an Int may be presumptuous:

***@***.***

Minor issue: You are probably aware if this but the node label is not stored:
***@***.***

NGDWalker, 2023-08-23:
Here is the error result from an IP write node: 
[image]

NGDWalker, 2023-08-23:
For debugging purposes, I added a "writeProperty" that loops through pointsToWrite and calls writeProperty for each, which does work all the way through to the device. So at least the pipes _are_ working.

I do like the discovery and "reading" functionality, particularly in a mixed IP/MSTP setup. The UI is very nice. It's worth spending some time on.

bitpool-dev, 2024-02-09:
Hi @NGDWalker 

Can you please update your module to the latest version and try again ? Removing all the nodes from a flow and restarting node-red would be ideal.

Thanks

bitpool-dev, 2024-05-16:
Closing due to inactivity, please try again with latest module version. If the issue is still present, come say hello in our discord or create a new issue.

closed 2024-05-16 by bitpool-dev