bitpool/edge-bacnet #11: cannot write some value

websystemsoftware, 2023-11-11:
the Product is very nice..
but it have some bugs i think..
i can read data without problems but it is not possible to write any Data!

and also some crazy stuff happend when i go with the standard settings
from time to time the Devices goes offline and come online, the complete devices are gone and so on...
after a time it was showing me 0 Devices, all devices are gone and come not back...
after i uninstall and install the edge-bacnet new in nodered it found all the devices again...

i think the problem comes when i test to write someting, and writing is not working....
when i was looking with yabe the devices was always perfectly online with no problems
only edge-bacnet have this problem was lost always the connection of devices...

i change now in the Gateway Node some settings in Device Discover Frequency and Object Discover Frequency to a higher value
and now look better > it lost no connections anymore at the moment....

**but the biggest problem is it is not possible to write some Values!!**

 gives here a solution to bring write a value working, or comes a update for this problen in the next time?
 
 Thanks for your nice work
 
 Benny

websystemsoftware, 2023-11-11:
[image]

the debug shows this after write...
but it writes not correctly the presentValue to the device...
when i change the value in yabe tool then it is working and i can change the presentValue from the yabe tool without problems

but when i write with edge-bacnet node nothing is write to the presentValue

davidjnblanch, 2023-11-11:
We are aware of the write issue and have a new release out soon, can you please provide details on the system and setup , is this bacnet mstp or ip ?

websystemsoftware, 2023-11-11:
[image]

Hello
we use NodeRed 3.1.0 and your plugin in Version 1.2.6
we use Bacnet IP
i send you also a printscreen out of yabe tool of our system

dblanchie, 2023-11-11:
> 
> 
> 
> 
> 
> 
> Hello
> 
> we use NodeRed 3.1.0 and your plugin in Version 1.2.6
> 
> we use Bacnet IP
> 
> i send you also a printscreen out of yabe tool of our system
> 
> 

No worries, is your node device running on windows or Linux, is the bacnet device bacnet ip or mstp ?

websystemsoftware, 2023-11-11:
we runing NodeRed on a Linux System
and the bacnet device is IP

dblanchie, 2023-11-11:
> we runing NodeRed on a Linux System
> 
> and the bacnet device is IP

Thanks can you provide device manufacturer and model, we will send you an update when the new release is available. We will also have ui improvements and bug fixes

websystemsoftware, 2023-11-11:
[image]

and this is our WriteList where i want to write the value > change the Present Value

ok i will check the device manufactur quick
the device > Bacnet IP Gateway was installing by a heating company...

dblanchie, 2023-11-11:
> [image]
> 
> 
> 
> and this is our WriteList where i want to write the value > change the Present Value
> 
> 
> 
> ok i will check the device manufactur quick
> 
> the device > Bacnet IP Gateway was installing by a heating company... 

I can't make out from the original picture but it looks like a Siemens MSTP router making this device an mstp device that you are writing to, maybe wrong hence why I asked for the model but it looks that way from the yabe image. We have a known issue with mstp writing, we have tested bacnet ip writing and have had any issues to date. If you can get the make and model that would be great, I will message back here when the new update will be released, shouldn't be too long

websystemsoftware, 2023-11-11:
[image]

helps you this printscreen with the Data?
i think is a Siemens PXG Router LON/Bacnet

but how i say this device installtion makes the heating company and iam only the IT Guy 
but when you need some more Infos about the device i must ask the heating company

Thanks
Benny

dblanchie, 2023-11-11:
> [image]
> 
> 
> 
> helps you this printscreen with the Data?
> 
> i think is a Siemens PXG Router LON/Bacnet
> 
> 
> 
> but how i say this device installtion makes the heating company and iam only the IT Guy 
> 
> but when you need some more Infos about the device i must ask the heating company
> 
> 
> 
> Thanks
> 
> Benny
> 
> 
> 
> 
> 
> 

Thanks man yep the device you are writing to is mstp which means it's a 2 wire rs485 device that sits under the Siemens ip gateway...this is a known issue for us so all good fix will be available soon

websystemsoftware, 2023-11-11:
thank you so much for your quick response and all of your work for this fantastic product!!!

i will wait for the Update

Greetings Benny

bitpool-dev, 2024-02-09:
Received confirmation that writing to points is now functional. 

Closing this issue.

closed 2024-02-09 by bitpool-dev