bitpool/edge-bacnet #12: Devices Present in Yabe, but not in Edge-BACnet Read Devices - Siemens PXG3.L BACnet to LonTalk router

HarvUK, 2023-11-22:
**Describe the bug**

First of all, many thanks for an absolutely amazing BACnet tool.

Only one device is listed to add in the read list (there are 7 on a windows install on the same network) - see the error log. I assume the file "edge-bacnet-datastore.cfg" may need creating or permissions changing on the OS.

[image]

`find . -type f -name "edge-bacnet-datastore.cfg" ` did not find the file present.

This issue does not happen on a MS windows install.

This is from the windows install
[image]

**To Reproduce**
Inject discovery into gateway

**Expected behavior**
All 7 devices present in read list.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information (please complete the following information):**
 - OS: [e.g. iOS] DietPi 8.24.1
 - Browser - Chrome
 - Version [e.g. 1.2] 1.2.6
 - Node-RED version:  3.1.0
 - NodeJS version: v21.2.0

**Are you running Node-RED in a docker container or directly on the operating system or virtual machine?**
Directly on the machine

**Please provide a dump of the Node-RED error log with the Error logging and Device Found check box options enabled (found in _gateway_ node - _Discovery_ tab) if applicable:**
 
Nov 22 12:29:15 DietPi node-red[1079]: 22 Nov 12:29:15 - [warn] [Bacnet-Gateway:NR Gateway] BACnet device found: 1050625 - 192.168.1.2
Nov 22 12:29:25 DietPi node-red[1079]: Store_Config writeFile error:  [Error: EACCES: permission denied, open 'edge-bacnet-datastore.cfg'] {
Nov 22 12:29:25 DietPi node-red[1079]:   errno: -13,
Nov 22 12:29:25 DietPi node-red[1079]:   code: 'EACCES',
Nov 22 12:29:25 DietPi node-red[1079]:   syscall: 'open',
Nov 22 12:29:25 DietPi node-red[1079]:   path: 'edge-bacnet-datastore.cfg'
Nov 22 12:29:25 DietPi node-red[1079]: }
Nov 22 12:29:35 DietPi node-red[1079]: Store_Config writeFile error:  [Error: EACCES: permission denied, open 'edge-bacnet-datastore.cfg'] {
Nov 22 12:29:35 DietPi node-red[1079]:   errno: -13,
Nov 22 12:29:35 DietPi node-red[1079]:   code: 'EACCES',
Nov 22 12:29:35 DietPi node-red[1079]:   syscall: 'open',
Nov 22 12:29:35 DietPi node-red[1079]:   path: 'edge-bacnet-datastore.cfg'
Nov 22 12:29:35 DietPi node-red[1079]: }

**Additional context**
Add any other context about the problem here.

bitpool-dev, 2023-11-23:
Hi @HarvUK 

Thanks for the feedback. 

This is an issue we have encountered previously, but only seems to be more present recently on linux based applications. 

Are you able to run the node-red service / process with elevated permissions ? Ideally with some form of permission that allows node-red to create files ? 

Some context behind the file: 
We use  'edge-bacnet-datastore.cfg' as a flat text file for storage of the known bacnet network, with the intention of a simple performance improvement mechanism when restarting node-red. Having a base structure when rebooting node-red pre-populates the UI rather than having to rediscover everything on restarts. 

As it is a recurring issue we will definitely address the permission issue on our next release.

HarvUK, 2023-11-23:
Many thanks, I'll have a see if I can find a way to make it work.

Having a flat file data store makes a lot of sense.

Harv

pangrazi, 2024-02-07:
I am hitting this same issue.  Can you tell me where the file is created?  I'll try adding a blank file with world rw perms to see if that works.

HarvUK, 2024-02-08:
Hi,

This week did a clean install of DietPI (v9.0.2) and installed the latest Node-Red (3.1.4) and the latest version of this module (1.2.8), and it all worked perfectly, with no issues, so I think this may have been resolved in the latest release of the module.

Sorry I don't have the location of the file, but hopefully this will help?

Thanks,

Harv

bitpool-dev, 2024-02-09:
Hi @pangrazi 

The file is created in the directory from which the node-red process / command is run. So it will depend on your install and how you run the program. Otherwise, there is a "Cache File Enabled" checkbox in the discovery tab of the Gateway node. Please disable this and all file related operations will be ignored. 

Hi @HarvUK 

Thats good to hear, glad its working smoothly for you now. 

Closing this issue due to the original problem being solved.

closed 2024-02-09 by bitpool-dev