bitpool/edge-bacnet #13: Buffer Overflow When Reading Device

HarvUK, 2023-11-22:
**Describe the bug**
After running a poll devices, the system gives errors on most of objects on the device - a total of 13 are imported of the 1554 objects on this device.

For further information this issue may be related to #11 as this system is also on a PXG3.L router, with the device being on a LonTalk network. Three of the other four devices on this router all read perfectly, with one other giving the same issues.

[image]

The device being interrogated is a PXC100.D
[image]

The other problem device is a PXC12.D with just 53 objects.

However, the other PXC12.D with 198 objects didn't give any errors at all.

**To Reproduce**
Steps to reproduce the behavior:
Suspect this is device specific, YABE exports available if required.

**Expected behavior**
Device is added into your brilliant UI and then we can add objects to be read.

**Screenshots**

This is the error as seen from wireshark.
[image]

When the read (eventually) completes, the device we are trying to work on has an exclamation next to it.

[image]

**System Information (please complete the following information):**
 - OS: Windows 11
 - Browser Chrome
 - Version 1.2.6
 - Node-RED version: 3.1.0
 - NodeJS version: 20.9.0

**Are you running Node-RED in a docker container or directly on the operating system or virtual machine?**
Directly on the machine

**Please provide a dump of the Node-RED error log with the Error logging and Device Found check box options enabled (found in _gateway_ node - _Discovery_ tab) if applicable:**
 
22 Nov 21:04:55 - [warn] [Bacnet-Gateway:NR Gateway] BACnet device found: 1050625 - 192.168.1.2
22 Nov 21:04:55 - [info] [mqtt-broker:Ventana] Connected to broker: mqtt://dashboard.hvasystems.co.uk:1883
22 Nov 21:04:56 - [warn] [Bacnet-Gateway:NR Gateway] BACnet device found: 2098177 - 192.168.1.2
BACnet Error:
Error while fetching objects: Error: BacnetAbort - Reason:1 undefined
BACnet Error:
Error getting point list for [object Object] - 2098177:  Error: BacnetAbort - Reason:1
    at Client._processAbort (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:102:40)
    at Client._handlePdu (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:411:22)
    at Client._handleNpdu (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:455:14)
    at Client._receiveData (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:467:18)
    at Transport.emit (node:events:514:28)
    at Socket.<anonymous> (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\transport.js:13:57)
    at Socket.emit (node:events:514:28)
    at UDP.onMessage [as onmessage] (node:dgram:941:8)
BACnet Error:
getDevicePointList error: [object Object] Error: BacnetAbort - Reason:1
    at Client._processAbort (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:102:40)
    at Client._handlePdu (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:411:22)
    at Client._handleNpdu (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:455:14)
    at Client._receiveData (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\client.js:467:18)
    at Transport.emit (node:events:514:28)
    at Socket.<anonymous> (C:\Users\stewa\.node-red\node_modules\@bitpoolos\edge-bacnet\resources\node-bacstack-ts\dist\lib\transport.js:13:57)
    at Socket.emit (node:events:514:28)
    at UDP.onMessage [as onmessage] (node:dgram:941:8)

bitpool-dev, 2023-11-23:
Hi @HarvUK 

Thanks for submitting this bug report. 

Due to the nature of the BACnet stack we've employed for our node-red applications, complete device coverage is still a work in progress. Its positive to hear that it covers most of your devices, but strange that another of the same working model isn't behaving correctly (PXC12.D)

Are the PXC12.D and PXC100.D child devices of the PXG3.L router ? Do they support segmentation at all ? 

A YABE and Wireshark dump would be very helpful, as seeing a buffer overflow abort is kind of odd. 

There are a few possible icons in the UI of a device list:
- Server icon - indicates a healthy bacnet/ip device

- Connected node icon - indicates a healthy bacnet/mstp device (Usually down stream from a router)

- Exclamation icon - indicates an error with the device Object List, and an activation of a manual discovery mode, hence the issue above.

Do the object lists for your problematic devices look correct in YABE etc ?

HarvUK, 2023-11-23:
Hi,

Thanks for the speedy response.

I've had another good test this morning, using the device ID filters to just request from one device at a time, and both PXC12.D devices seem to read OK, this may well have been an operator error - sorry.

Both the PXG3.L and the PXC100.D seem to have segmentation enabled (screenshots from YABE):

[image]

PXC100.D
[image]

The objects all look good in YABE, attached below are YABE EDE files and Wireshark dumps from a PCX100.D (device 2098177) and a PXC12.D (device 2095179).

I've also included the command line logs from the run which fails on the PXC100.D.

If there's anything else I can do or provide to assist, please let me know, and thank you once again.

[Files for bitpool v1.zip](https://github.com/bitpool/edge-bacnet/files/13449751/Files.for.bitpool.v1.zip)

Harv

bitpool-dev, 2024-02-09:
Hi @HarvUK 

Can you please update your module to the latest version and try again ? Removing all the nodes from a flow and restarting node-red would be ideal.

Thanks

HarvUK, 2024-02-09:
> Hi @HarvUK
> 
> Can you please update your module to the latest version and try again ? Removing all the nodes from a flow and restarting node-red would be ideal.
> 
> Thanks

Hi,

Good news, I removed the module and updated node red to the latest version (3.1.5), and then reinstalled your module and started again, and success - it worked perfectly!

Many thanks for your work on this!

Harv

dblanchie, 2024-02-09:
> Hi,
>
>
>
> Good news, I removed the module and updated node red to the latest version (3.1.5), and then reinstalled your module and started again, and success - it worked perfectly!
>
>
>
> Many thanks for [...]

We will look to update the docs to with some process around updates too, we have just been behind with doing updates needed to get the connector right.

bitpool-dev, 2024-02-12:
Great to hear ! Closing this issue as it is fixed.

closed 2024-02-12 by bitpool-dev