bitpool/edge-bacnet #15: Discovery not able save

m2b, 2024-02-03:
**Describe the bug**
Discovery works fine but it cannot save due to config file not existing to begin with.  Had to manually place and empty file in the directory to be able to save.

**To Reproduce**
Steps to reproduce the behavior:
1. Running on Windows.
2. No Docker container.  Just node-red.
3. Import Read-Discover example.
5. Trigger discovery.
6. Errors visible in Node-red console or log file.
Store_Config writeAccess error found:  [Error: ENOENT: no such file or directory, access 'C:\Users\NAME\Documents\sandbox\edge-bacnet-datastore.cfg'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'access',
  path: 'C:\\Users\\NAME\\Documents\\sandbox\\edge-bacnet-datastore.cfg'

**Expected behavior**
Should create file if not exists.  At least in Windows, fs.access(W_OK) returns error if file not there.

**Screenshots**
This shows discovery works.
[image]

**System Information (please complete the following information):**
 - Windows11 Pro
 - Node-RED version:  v3.1.3
 - NodeJS version:  v20.11.0

**Are you running Node-RED in a docker container or directly on the operating system or virtual machine?**
No.  Just on the OS.

**Please provide a dump of the Node-RED error log with the Error logging and Device Found check box options enabled (found in _gateway_ node - _Discovery_ tab) if applicable:**
 See error above.  Output was in console.

**Additional context**
Add any other context about the problem here.

dblanchie, 2024-02-03:
Known issue and will be update in next release shortly

bitpool-dev, 2024-02-09:
Hi @m2b  

Can you please update your module to the latest version and try again ? There is a Cache File Enabled check box in the Discovery tab of the gateway node. Please disable this if your file is still unable to be created. 

Thanks

bitpool-dev, 2024-05-16:
Closing due to inactivity, please try again with latest module version. If the issue is still present, come say hello in our discord or create a new issue.

closed 2024-05-16 by bitpool-dev