bitpool/edge-bacnet #16: Read Node not showing devices

m2b, 2024-02-03:
Only shows count.  No errors in the log file.   I was able to have discovery save the config (see other bug reported today).  I can even export it or import config.  

OS: Windows11 Pro
Browser: Edge
[image]

dblanchie, 2024-02-03:
This is a known issue and will there will be an update release shortly

bitpool-dev, 2024-02-09:
Hi @m2b 

Can you please update your module to the latest version and try again ? Removing all the nodes from a flow and restarting node-red would be ideal. 

Thanks !

m2b, 2024-02-10:
I upgrade and have been testing.  
1. So far, no exception on discovery. But where is it saving saving the config file?  I could not locate it.
2. It is however, showing 3 less devices on discovery than before.   
4. As far as read, same problem.  Shows only count.  One time it showed on device with a count of 7 (there are 10 devices that Yabe can see). At the next discovery the device disappeared.
5. I am investigating if there is an issue since other application on the same host is reading the devices correctly.  
6. I am working on setting up a separate host to run node-red and then try again.

My requirement is not discovery.  My limited experience is that it does not always work.  I have had device issues where rebooting the device is necessary to restart discovery.   Even Yabe has had issues discovering.    The other issue with discovery has been with Routers.  They don't always show their attached MSTP devices on discovery.

My preference is to have the device vendor to provide me a list of items in the device and then just setup the Bacnet read to use read those points.  Any chance that you can setup read to not depend on a discovered devices list?

In case you are interested the devices I am testing are Moxa gateway (Modbus to Bacnet), Accuenergy Power Meters, Flexim Flow meters, and JACE Controllers setup as a routers to Bacnet/MSTP Flexim Flow Meters.

davidjnblanch, 2024-02-10:
ok, just a few quick points, make sure you don't have 2 gateway nodes on the same flow, the gateway is to be common and you can have multiple read and write nodes into the same gateway. You also can't run node bacnet on the same device that another BACnet application is running on

bitpool-dev, 2024-05-16:
Hi @m2b  - could you please try again with the latest module version ? Please follow the updating steps in the readme if possible.

bitpool-dev, 2024-08-01:
Please try again with latest version. 
Closing this issue due to inactivity, it can be re opened at any time or a new issue created alternatively.

closed 2024-08-01 by bitpool-dev