bitpool/edge-bacnet #17: MST Device Not selectable in Read or Write list

pangrazi, 2024-02-19:
In the read or write device list, my BACnet router and MSTP network both show.  I am able to hit the "plus" and add the BACnet router points, however the MSTP points cannot be added.  I found a work-around that if I click "add all" on the read I will get the data from the read side, however I still cannot remove individual points or add points to write to.

[image]

[image]

[image]

There do not seem to be any log messages associated with clicking the "plus" to add a device.

OS: Linux kernel 5.15 Raspberry Pi
Browser Firefox
Version 1.2.8
Node Red Version v3.0.2
Node JS version 16.20.2

Running Directly on the machine no vm or container.

davidjnblanch, 2024-02-19:
thanks for this, we have fixed this in the next release that is coming shortly. thanks for your patience

pangrazi, 2024-02-19:
Great!

Let me know if you want me to test some code.

Thanks,
Greg Pangrazio
***@***.***
847-707-7933 (c)
AC9AF

pangrazi, 2024-03-11:
Any luck on this issue yet?

pangrazi, 2024-04-18:
How is it going on this release?  Any way I can help?

pangrazi, 2024-05-19:
I don't think the updates helped.

[image]

The Device count says "3" but it never finds the MST device

The write node never finds the MST device at all now.

[image]

I tried rebuilding the data model and reloading the data with no effect.  What data can I provide to help troubleshoot?

pangrazi, 2024-05-19:
After posting this, I did another restart of node-red on my system.  Now the values show on the read list, but not properly.  They are under the wrong device - these should show under ID 124 not 1000

[image]

Additionally, I can't get any of the points to delete when I click the minus button.  If I click "remove all" the points get deleted, and the only way to get them back is a full restart of node-red.

bitpool-dev, 2024-05-20:
hi @pangrazi can you please send a screenshot of your gateway node properties? network adapter, broadcast address etc.. Also could you take a screenshot of the devices using YABE?

If these two things are looking right, a 1min wireshark capture of node-red starting would also help alot.

pangrazi, 2024-05-20:
[image]
[image]
Device 124 is the MSTP device, Device 1000 is a BACNET ethernet router from Chipkin.

pangrazi, 2024-05-20:
[udp47808.zip](https://github.com/bitpool/edge-bacnet/files/15372126/udp47808.zip)
Here is a capture of the traffic generated by my system.  The IP is 192.168.1.206 of the PI running Node-Red and the libraries.

bitpool-dev, 2024-05-20:
Thanks for that, will take a look at the capture

I noticed the network interface is empty in your gateway properties, can you select an interface, deploy node-red, then restart it? 

Another question, are you running multiple bacnet services on the same network interface / ip address? In my experience this can be problematic

pangrazi, 2024-05-20:
I have set the IP address to "all interfaces" per your doc.  Deployed and restarted.  The host running Node Red only has one bacnet service on it.  The bacnet router has itself and the MSTP based furnace.  That did not seem to make any change to the config.  I also turned off the cache file as we had problems in the past.

pangrazi, 2024-05-20:
I also enabled logging of found devices and when I "discover" the gateway node reports finding the furnace device, however it still doesn't show properly in read/write nodes.