bitpool/edge-bacnet #18: What do the icons mean, and what determines which icon is shown?

CV-Private, 2024-03-04:
Hello Bitpool, thanks for this project! 

The discovery of my Brötje heatpump went fine. All of the 1321 objects were found. The Read node works fine. I'm still fighting with the Write node and I have currently the same problem as reported in issue #14. This issue doesn't seem to be fixed yet.

In this context, I'm wondering what the icons mean. All of my data points have a gearwheel icon and all objects have the (R) symbol.
In your videos and documentation, I'm seeing a lot of other icons, though.
Is there any relation between the icons and the read/write properties?

This is what I'm seeing:

[image]

This is from one of your videos:

[image]

System Information:
 - BACnet Device: Brötje GTW-21 (BDR Thermea Group), BACnet certified 
 - Browser: Firefox
 - Node-RED version: 3.1.6
 - Bitpoolos/edge-bacnet version: 1.2.8

I'm running Node-RED on a VM on Proxmox. I didn't use the Docker container.

davidjnblanch, 2024-03-04:
hey there, so we are behind in the video :) we changed the icons to try and make more sense. In the next release, we will be updating documentation and videos. (R) stands for read only and (W) stands for read and write. Write capability is set by the OEM device and this is determined by the existence of a write priority array being available. fyi gear icon stands for property of point.

CV-Private, 2024-03-05:
Thanks, David! 
Regarding the (R) and (W): that is what I already suspected.
 
Well, with this being said, there seems to be a bug in determining the read/write property. My GTW-21 has plenty of data points that can be written, but I don't see any (W) in the device list.

I have already tested and confirmed that I can write to the AI-2126 object from Yabe. However, object varZoneTRoomSetpoint-3_AI_2126 is shown as (R) in the node despite being a writable Setpoint:

[image]

Here's the same AI_2126 object, shown by Yabe:

[image]

davidjnblanch, 2024-03-05:
ok i see the issue, so this comes down to interpretation,  writing to an analog input, i.e a physical input on the device would not be considered "good practice" I'm sure there may be plenty who would disagree, but the principle is that if this was a physical input on the control you would have a remote offset adjuster or something controlling this physically. So with this being said we look for an object with a priority arrary for writing (1-16). Im assuming that in yabe you are just adjusting the presentValue. The other key reason we do this is you will get circumstances where yabe would error out if the OEM device doesnt allow writing hence the priority array is used to determine this, if a priority arrary exists its definalty writable. This maybe not be the answer you are after :) its just the logic we used when developing the connector

CV-Private, 2024-03-05:
Hello David, I'm by no means an expert in the BACnet subject and I wouldn't dare to argue with you on how to interpret the standard.

However, my GTW-21 BACnet module is made by BDR Thermea (www.bdrthermeagroup.com) and came with a bunch of BACnet certificates. Therefore I'd assume that it has been developed and tested in accordance with the official BACnet specification.

The physical input on my heatpump is the control panel. I can directly set the value varZoneTRoomSetpoint on the control panel and I see the effect in Yabe immediately. I would interpret that as a valid Analog Input to the device with me - the user - adjusting the setpoint input.

So, no matter on who's interpretation is right or wrong, fact is that the Bitpool Connector is currently not fully interoperable with the GTW-21 device, and I have no whatsoever way of changing anything in the GTW-21 BACnet device.

I haven't yet been able to fully integrate my GTW-21 in my home control system. None of the available BACnet interfaces are fully interoperable with that device yet. Only Yabe allows me to access it without any flaws, but Yabe is not meant to be a Connector.

I'd really love to get the Bitpool connector working with my device, because I like the functionality a lot. I'm hoping that Bitpool would aim to improve and extend the interoperabilty of the Connector, because interoperability is the whole point of the BACnet idea.

Let me know if I can assist in any way, in case that you'd like to further work on this issue.

davidjnblanch, 2024-03-05:
all good, I need to do some research, interestingly we previously had the ability , the problem as i described above was that some AI are writable and some are not, when they are not how is the error handled. Information from the standard below, we are trying to stay with this however has even described in the document we dont necessarly have to :)

[image]

[image]

[image]

CV-Private, 2024-03-05:
Just thinking out loud: maybe Brötje defined the object varZoneTRoomSetpoint-3_AI_2126 as Analog Input because it can also be adjusted remotely from a control panel in the room:

[image]

So, the room controller would be considered being a "sensor" that senses the user input and provides it to the heat pump controller? 

Anyway - the GTW-21 also has a lot of adjustable parameters, such as parZoneDhwAntilegionelSetpoint-3_AV_1692:

[image]

Brötje has chosen to make this object an Analog Value, and it can be configured on the control panel of the heatpump.

However, the Bitpool Connector also shows it as read-only (R), as you can see in the screenshot above.

CV-Private, 2024-03-06:
Hello Daniel, I have an update after digging further into the GTW-21 documentation. 

Parameter parZoneRoomManualSetpoint-3_AV_1688 is the value that comes from my temperature-control-panel in the room. It's an Read/Write Analog Value:

[image]

This is how this object looks like in Yabe:

[image]

This is how the object looks like in the Bitpool Write node:

[image]

So, there's apparently an inconsistency in the identification of the Read/Write property of this object between the Bitpool Write node and the intended purpose of the object.
Writing to this object from the Write node fails.

The GTW-21 came with a PICS (Protocol Implementation Conformance Statement) and is BSL Listed, therefore I would expect that BDR has implemented the protocol in accordance with the BACnet standard:

[image]

dblanchie, 2024-03-06:
Hi There,

As i described above, **Bitpool identifies a point that has a write priority array as writable**, the point you show above appears not to have a priority array hence we don't identify it as writable. The point could very well be writable by placing it into out of service and commanding a value however we don't support this method.

CV-Private, 2024-03-07:
Ok, understood. Pity, I will then have to search further for an alternative solution that allows me to write to my BACnet device, as there doesn't seem to be any way for establishing full interoperability between my GTW-21 and the current Bitpool version.
Anyway - thanks for your support and for identifying the root cause of the problem!
Best regards, Christian

closed 2024-03-07 by CV-Private