bitpool/edge-bacnet #19: Why doesn't Node-RED find the Yabe Room Simulator?

CV-Private, 2024-03-04:
Why can't Node-RED find the Yabe Room Simulator Device 1507482? 

[image]

Yabe does find it: (Device 1 is the GTW-21 and Device 420 is the Eco-Panel.)

[image]

I have no clue, though, why the Room Simulator appears on Port 63814. I didn't configure it like that (I don't even know how to...).

I tried already to set the Port in the Gateway Node to 63814, but that didn't have any effect - Node-Red can still only find 2 of the three BACnet devices.

System Information:
BACnet Device: Brötje GTW-21 (BDR Thermea Group), BACnet certified
Browser: Firefox
Node-RED version: 3.1.6
Bitpoolos/edge-bacnet version: 1.2.8

I'm running Node-RED on a VM on Proxmox. I didn't use the Docker container.

CV-Private, 2024-03-05:
Hi David, thanks for your swift response!
Indeed, the room simulator runs on the same machine. I'll try to figure out if I can solve that in my firewall settings.

bitpool-dev, 2024-05-16:
Hey @CV-Private  - is this still an issue ? Could you please try again with latest module version, following the updating steps in readme if possible

bitpool-dev, 2024-08-01:
Please try again with latest version.
Closing this issue due to inactivity, it can be re opened at any time or a new issue created alternatively.

closed 2024-08-01 by bitpool-dev