bitpool/edge-bacnet #20: Read list doesn't seem to be updated properly

CV-Private, 2024-03-04:
The Read Node of my flow polls a data point (Algorithm Type) that I had removed from the Read List several deployments ago. Another value in the Read List (varZoneTRoomSetpoint-3_AI_2126) is not getting polled.
Could there be an issue in updating the Read List?

See the screenshot below:

[image]

System Information:

- BACnet Device: Brötje GTW-21 (BDR Thermea Group), BACnet certified
- Browser: Firefox
- Node-RED version: 3.1.6
- Bitpoolos/edge-bacnet version: 1.2.8

I'm running Node-RED on a VM on Proxmox. I didn't use the Docker container.

davidjnblanch, 2024-03-05:
send through screen capture of AI_2126 expanded with properties

CV-Private, 2024-03-05:
Hello David, please see below the expanded view of the AI_2126 with properties.

As a further observation I noticed that all of the "Present Values" in the Read List are old. The payload values are correct and current, though. In the screenshot below, I drew lines between the data points and their corresponding Payload values in the Debug output.

[image]

When I look at the Device List, then I see different Present Values, which are neither equal to the current payload values nor to the Present Values in the Read List.

In the Device List, the GTW-21 is shown Online. In the Read List, there's no such indicator. Should there be one as well?

[image]

Note that I rebooted Node-Red this morning. That didn't make any change to the behavior of the Read node, nor to the Read List.

davidjnblanch, 2024-03-05:
so there are a few things here, on the gateway node can you send a screen of the setting, for the object discovery this should be set to your desired update frequency. This behavior is not the typical experience a user would expect and has been fixed in our next release.

davidjnblanch, 2024-03-05:
> Hello David, please see below the expanded view of the AI_2126 with properties.
>
> As a further observation I noticed that all of the "Present Values" in the Read List are old. The payload values are [...]

in relation to the status indicator, no there is not one on the read list, the read list is simply the points to be polled by each request and the intnet is the device list shows the status of the device

CV-Private, 2024-03-05:
Indeed, the Gateway node has 1 minute frequency settings, so it appears that the errors in the log are related to the Discovery settings.

[image]

My Inject node is currently polling the Read List every 5 seconds, which doesn't seem to cause any errors in the log. I haven't yet found any way for configuring the Read node to COV (Change of Value), which would be very desirable. 

[image]

bitpool-dev, 2024-05-16:
hey @CV-Private - the front end code for adding and removing points from the device list to the read list can be a bit painful. If points are not removing from the read list at all, its most likely because the name has changed for the point in the device object or that point no longer exists. Im hoping to have fixed this with the latest node version. 

If the points do not want to be removed at all, please delete the read node entirely, deploy node-red, and replace with a new read node. 

Could you please update to latest node version and try again ? Please follow the updating instructions on the readme if possible.