bitpool/edge-bacnet #21: Couldn't read 'Object List' count on YABE

UnCanard14, 2024-05-15:
Hello, thank you very much for your tools.
I have a problem when I make the flow by default:
[image]
[image]

I get this error when I try to access the objects on YABE:
[image]

I modified the line :
https://github.com/bitpool/edge-bacnet/blob/d01ffdb897768fb27a83cbd55d8e1e026f6cacef/bacnet_server.js#L158
by :
`that.bacnetClient.client.readPropertyResponse(data.address, data.invokeId, data.request.objectId, data.request.property, responseObj);`

And on YABE I have no more problems... How can we fix this?

Translated with DeepL.com (free version)

dblanchie, 2024-05-15:
> Hello, thank you very much for your tools. I have a problem when I make the flow by default: ![image-1](https://private-user-images.githubusercontent.com/25232448/330745672-c130dacf-2123-4900-b317-432115fb8332.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3NjU3NjYsIm5iZiI6MTcxNTc2NTQ2NiwicGF0aCI6Ii8yNTIzMjQ0OC8zMzA3NDU2NzItYzEzMGRhY2YtMjEyMy00OTAwLWIzMTctNDMyMTE1ZmI4MzMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDA5MzEwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0NjY2Y2MzYTAyMWZkMzJhMTY3ZTE4MGFjYWFlY2M4ODJhODc3ZmY5NzIwY2YzZmJlNWI0MjE0YjgzNGEyMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WoxQ7u9GyVWUjjcdDprSf7CYsgfLmZwL9YP56Z_lSZ4) ![image](https://private-user-images.githubusercontent.com/25232448/330745605-d6c82326-4851-4f4e-bd00-35a8f7d6a9b5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3NjU3NjYsIm5iZiI6MTcxNTc2NTQ2NiwicGF0aCI6Ii8yNTIzMjQ0OC8zMzA3NDU2MDUtZDZjODIzMjYtNDg1MS00ZjRlLWJkMDAtMzVhOGY3ZDZhOWI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDA5MzEwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4Y2Y2YjUxN2RjMmFlYWU1NGYyMDc3OTY4YzM1MWE2ZGVkM2YwMGE4YTg2ZWMzYmRmMzYxNDMyMTdmYTc2YjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wpsALqMUyM6zozGakc5n0SDGuxKR6wvsjS0ZkgRry3A)
> 
> I get this error when I try to access the objects on YABE: ![Capture d'écran 2024-05-15 103612](https://private-user-images.githubusercontent.com/25232448/330745756-2265ebea-a165-4107-aafe-4f6c1551e544.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3NjU3NjYsIm5iZiI6MTcxNTc2NTQ2NiwicGF0aCI6Ii8yNTIzMjQ0OC8zMzA3NDU3NTYtMjI2NWViZWEtYTE2NS00MTA3LWFhZmUtNGY2YzE1NTFlNTQ0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE1VDA5MzEwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxODM4ZDlmNTU3MTc3MjQ5NDgwY2UwZGQ1ZDM2Y2U1NmYxNmFhMjVkOTBlMGQyMTM4ZjNlYzlhNjk0M2ZmZjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DRA0MC03tJmVoD6BlGPMo07LL05fn1QQdxUjRftZ8Bc)
> 
> I modified the line :
> 
> https://github.com/bitpool/edge-bacnet/blob/d01ffdb897768fb27a83cbd55d8e1e026f6cacef/bacnet_server.js#L158
> 
> 
> by :
> `that.bacnetClient.client.readPropertyResponse(data.address, data.invokeId, data.request.objectId, data.request.property, responseObj);`
> And on YABE I have no more problems... How can we fix this?
> 
> Translated with DeepL.com (free version)

hi there

dblanchie, 2024-05-15:
Hi There,

We are doing a major release tommorow that fixs many issues with this node along many perfromance improvments. This issue is resolved in tommrows release

UnCanard14, 2024-05-15:
Oh great thank you so much !!!!

UnCanard14, 2024-05-16:
Hello, I've updated your node but I'm getting a new error when I try to collect my items 😅
[image]

bitpool-dev, 2024-05-17:
Hi @UnCanard14  - that is a bit odd. I can discover the bacnet node and see the points in Yabe on my end. Are you able to provide me with some additional information? 

OS: [e.g. iOS]
Node-RED version:
NodeJS version:
Are you running Node-RED in a docker container or directly on the operating system or virtual machine?

Also finally a Wireshark Capture for this error would be very helpful if possible.

UnCanard14, 2024-05-17:
Yes sure :
Os : Ubuntu 22.04.4 LTS
Node-red : 3.1.9
Node-js : v18.20.2
Node red is directly on the operating system with this setup : https://nodered.org/docs/getting-started/raspberrypi

debraysimon, 2024-06-03:
I @UnCanard14 I have the same problem as you, have you found a solution?

UnCanard14, 2024-06-03:
Hello @debraysimon sorry not yet but I contacted @dblanchie and he told me that there would be an update soon. 😉

bitpool-dev, 2024-06-06:
hey @UnCanard14 @debraysimon  - Can you update to version 1.3.1 and please try again ? 

Turns out I was using an older version of YABE that was compatible. Ive since updated the code and is working with latest version of YABE on my end.

debraysimon, 2024-06-06:
Great, it works for me ! Thank you !

UnCanard14, 2024-06-06:
Hello, it works for me too, thanks a lot !!

closed 2024-06-06 by UnCanard14