bitpool/edge-bacnet #22: Create a sub-device

UnCanard14, 2024-05-15:
I'm relatively new to the Bacnet protocol but I was wondering if there's a way to do something like this :
[image]
 with your node, or is specific equipment required?

dblanchie, 2024-05-15:
these are MSTP bacnet device which are rs485 2 wire devices, we support discovery of these devices. for BACnet server this can only be done from the IP device you are running node on, we dont support BACnet server over MSTP. Also note you will see a link to our discord which maybe better for you for flowing chat :)

UnCanard14, 2024-05-16:
Hello it works thank you for your answer, it enlightened me! Ah yes, that would be nice, where can I find this link? :)

bitpool-dev, 2024-05-17:
Hey @UnCanard14 - yes there are many BACnet MS/TP devices available from various manufacturers, happy shopping ! 

You can join our discord here if you are interested:    https://discord.gg/G4NFaMtqga

Closing this issue.

closed 2024-05-17 by bitpool-dev