bitpool/edge-bacnet #26: Cannot discover devices

doubleopinter, 2024-06-20:
Hi, I'm trying to use the driver and am having issues discovering devices. Details
- Nodered 3.1.11 running in a container, network is configured for host.
- OS is Ubuntu 20.04.
- I have two network interfaces on the device, both of which are connected to networks. One network is connected to the internet and the other one is connected to an internal network with my bacnet devices.

The device discovers itself but nothing else. I have another software (skyspark) that I also run in a container on that device and it is able to discover the devices (I don't run skyspark and node red at the same time). I've tried setting the network interface to every option but still nothing. 

Please let me know what else I can provide.

dblanchie, 2024-06-20:
Hi there,

Send a screen capture of the adapter list dropped down and your entire network settings view.

Also for discussion it is good to jump into the discord

Thanks

doubleopinter, 2024-06-20:
Hello, I'll send it asap. What network settings view are you referring to?

dblanchie, 2024-06-20:
The ones in nodered, on the gateway node

doubleopinter, 2024-06-20:
[image]
[image]
[image]

doubleopinter, 2024-06-20:
Just as a quick reference this is what Skyspark finds. I'm not comparing or expecting this to be skyspark, just as a sanity check.
[image]

bitpool-dev, 2024-06-21:
Hi @doubleopinter - Are you able to try and change the broadcast address on the gateway node to "10.254.251.255" while leaving the network interface option as "0.0.0.0" ? Please restart node-red after deploying this change

Also can you please confirm that you are not running multiple BACnet services on the same device concurrently ? Port 47808 does not like sharing network traffic, especially with node-red

doubleopinter, 2024-06-21:
Hi @bitpool-dev that worked! It instantly found the devices. Thank you. 
Related question. If we were in a situation where both interfaces had bacnet devices on them, would it work to scan both of those individually and then reading would still work on both trunks?