bitpool/edge-bacnet #31: Description fields not read into read list any more

HarvUK, 2024-07-16:
**Describe the bug**
In newer version of the module, the description field is not present in the read list.

Example of existing install
[image]

Example from new install
[image]

The new install is clean install of v1.4.1

**To Reproduce**
Steps to reproduce the behavior:
Install, discover devices

**Expected behavior**

Description fields to be present.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information (please complete the following information):**
 - OS: Raspberry Pi OS
 - Browser Chromium
 - Version 1.4.1
 - Node-RED version: 4.0.2 
 - NodeJS version: 

**Are you running Node-RED in a docker container or directly on the operating system or virtual machine?**
On Pi

**Please provide a dump of the Node-RED error log with the Error logging and Device Found check box options enabled (found in _gateway_ node - _Discovery_ tab) if applicable:**
 

**Additional context**

HarvUK, 2024-07-17:
Update:

I have looked on the device again today, and it appears that the description field is now updated.

I don't know why though - can the dev explain please - did I just need to wait for a points discovery?

[image]

bitpool-dev, 2024-07-18:
Hi @HarvUK   -  did I just need to wait for a points discovery? -  yes.

The point discovery flow has been adjusted to do an initial iteration of all points, only retrieving objectName and presentValue. After which it will backfill with additional properties. Reason for this is we have had sites with 500+ devices, some of which have 20k+ points. So this adjustment makes for a more responsive tree build out. 

If you want all of the information sooner, you can reduce the Object Discovery Frequency, or manual Right Click -> Update Points  on your devices.

HarvUK, 2024-07-18:
This makes sense - many thanks.

closed 2024-07-23 by bitpool-dev